osrf/vehicle_gateway #7: Not able to build on ros2 foxy

Jaeyoung-Lim, 2023-01-04:
**Problem Description**
I tried building this package on ros2 foxy (on Ubuntu 20.04), but was not able to due to the following issue

I tried to install `gz-cmake` but was not able to find where it would be distributed.

@ahcorde would you be able to provide any hints?
```
$ colcon build
Starting >>> gz-msgs9
Starting >>> sdformat13
Starting >>> ros_gz_interfaces
Starting >>> px4_msgs
Starting >>> vehicle_gateway
--- stderr: sdformat13                                        
CMake Error at CMakeLists.txt:21 (find_package):
  By not providing "Findgz-cmake3.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "gz-cmake3", but CMake did not find one.

  Could not find a package configuration file provided by "gz-cmake3" with
  any of the following names:

    gz-cmake3Config.cmake
    gz-cmake3-config.cmake

  Add the installation prefix of "gz-cmake3" to CMAKE_PREFIX_PATH or set
  "gz-cmake3_DIR" to a directory containing one of the above files.  If
  "gz-cmake3" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< sdformat13 [0.46s, exited with code 1]
Aborted  <<< gz-msgs9 [0.49s]
Aborted  <<< qgroundcontrol [0.61s]              
Aborted  <<< vehicle_gateway [0.66s]             
Aborted  <<< ros_gz_interfaces [2.47s]           
Aborted  <<< px4_msgs [2.90s]                                 

Summary: 0 packages finished [3.15s]
  1 package failed: sdformat13
  5 packages aborted: gz-msgs9 px4_msgs qgroundcontrol ros_gz_interfaces vehicle_gateway
  3 packages had stderr output: gz-msgs9 qgroundcontrol sdformat13
  11 packages not processed

```

codebot, 2023-01-04:
This project is targeting Ubuntu 22.04, ROS 2 Humble, and Gazebo Garden. Because of the number of internal and external dependencies, it's unlikely that it will work on earlier versions of those projects.

Jaeyoung-Lim, 2023-01-04:
@codebot Thanks for the pointers

I will try with ROS Humble in that case.

codebot, 2023-01-05:
We just merged some additional documentation into the README that describes how to set up the ROS 2, Gazebo, and PX4 builds. I'll close this issue for now, but feel free to re-open or open a new issue if the documentation doesn't work for you. Note that this project is under heavy construction :construction:  and will be pretty volatile for a while :construction_worker:

closed 2023-01-05 by codebot