osrf/vehicle_gateway #16: Build error with px4_sim

Jaeyoung-Lim, 2023-01-14:
**Problem Description**
I have followed the documentation on ROS Humble 22.04, and the build fails on building `px4_sim`

```
make[3]: warning: -j0 forced in submake: resetting jobserver mode.
-- PX4 version: v1.14.0-beta1-142-gedc531913d
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.6", minimum required is "3") 
-- PX4 config file: /home/jaeyoung/vg/vg_ws/build/px4_sim/px4-src/src/px4-firmware/boards/px4/sitl/default.px4board
-- PLATFORM posix                                               
-- ROMFSROOT px4fmu_common
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is GNU 11.3.0
-- The C compiler identification is GNU 11.3.0
-- The ASM compiler identification is GNU                       
-- Found assembler: /usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done                         
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: RelWithDebInfo
-- Building for code coverage
-- Looking for gz-transport12 -- found version 12.1.0           
-- Searching for dependencies of gz-transport12
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found suitable version "3.12.4", minimum required is "3") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.4
-- Found ZeroMQ: TRUE (Required is at least version "4")        
-- Checking for module 'uuid'
--   Found uuid, version 2.37.2
-- Found UUID: TRUE  
-- Looking for gz-utils2 -- found version 2.0.0
-- Searching for dependencies of gz-utils2
-- Searching for <gz-utils2> component [cli]
-- Looking for gz-utils2-cli -- found version 2.0.0
-- Searching for dependencies of gz-utils2-cli
-- Looking for gz-msgs9 -- found version 9.2.0
-- Searching for dependencies of gz-msgs9
-- Looking for gz-math7 -- found version 7.1.0                  
-- Searching for dependencies of gz-math7
-- Looking for gz-utils2 -- found version 2.0.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- ROMFS: ROMFS/px4fmu_common                                   
-- Configuring done                                             
-- Generating done                                              
-- Build files have been written to: /home/jaeyoung/vg/vg_ws/build/px4_sim/px4-src/src/px4-firmware/build/px4_sitl_default
[  1%] Building CXX object boards/px4/sitl/src/CMakeFiles/drivers_board.dir/board_shutdown.cpp.o
[  1%] Generating Flight Tasks
[  1%] Generating combined event json file
[  1%] Generating uORB topic headers
Failed to import genmsg: No module named 'genmsg'

You may need to install it using:
    pip3 install --user pyros-genmsg

gmake[6]: *** [msg/CMakeFiles/uorb_headers.dir/build.make:441: uORB/topics/action_request.h] Error 1
gmake[5]: *** [CMakeFiles/Makefile2:5371: msg/CMakeFiles/uorb_headers.dir/all] Error 2
gmake[5]: *** Waiting for unfinished jobs....
[  2%] Generating px4 event header file
[... 74 lines cut ...]
make[3]: warning: -j0 forced in submake: resetting jobserver mode.
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

make[3]: warning: -j0 forced in submake: resetting jobserver mode.
gmake[6]: *** [msg/CMakeFiles/uorb_headers.dir/build.make:441: uORB/topics/action_request.h] Error 1
gmake[5]: *** [CMakeFiles/Makefile2:5371: msg/CMakeFiles/uorb_headers.dir/all] Error 2
gmake[5]: *** Waiting for unfinished jobs....
gmake[4]: *** [Makefile:156: all] Error 2
make[3]: *** [Makefile:227: px4_sitl_default] Error 2
gmake[2]: *** [CMakeFiles/px4-firmware.dir/build.make:86: px4-src/src/px4-firmware-stamp/px4-firmware-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/px4-firmware.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< px4_sim [4min 33s, exited with code 2]

Summary: 28 packages finished [17min 58s]
  1 package failed: px4_sim
  2 packages had stderr output: gz-sim7 px4_sim

```

Jaeyoung-Lim, 2023-01-14:
Didn't identify exactly what was missing from the previous colcon build, but running the setup script from PX4 (https://github.com/PX4/PX4-Autopilot/blob/main/Tools/setup/ubuntu.sh) seems to install additional dependencies that are missing

Jaeyoung-Lim, 2023-01-14:
Unable to run the launchfile, as it results in a error

```
$ ros2 launch px4_sim x500_launch.py
[INFO] [launch]: All log files can be found below /home/jaeyoung/.ros/log/2023-01-14-15-27-01-715220-jaeyoung-ThinkPad-X1-Carbon-Gen-10-83776
[INFO] [launch]: Default logging verbosity is set to INFO
using rootfs  /tmp/tmp1cmsk87v
seeding rootfs at /tmp/tmp1cmsk87v from /home/jaeyoung/vg/vg_ws/install/px4_sim/share/px4_sim
[INFO] [ruby $(which gz) sim-1]: process started with pid [83778]
[INFO] [px4-2]: process started with pid [83780]
[INFO] [QGroundControl.AppImage-3]: process started with pid [83783]
[INFO] [micro_ros_agent-4]: process started with pid [83786]
[ERROR] [QGroundControl.AppImage-3]: process has died [pid 83783, exit code 1, cmd 'QGroundControl.AppImage'].
[QGroundControl.AppImage-3] dlopen(): error loading libfuse.so.2
[QGroundControl.AppImage-3] 
[QGroundControl.AppImage-3] AppImages require FUSE to run. 
[QGroundControl.AppImage-3] You might still be able to extract the contents of this AppImage 
[QGroundControl.AppImage-3] if you run it with the --appimage-extract option. 
[QGroundControl.AppImage-3] See https://github.com/AppImage/AppImageKit/wiki/FUSE 
[QGroundControl.AppImage-3] for more information
[px4-2] INFO  [px4] instance: 0
[px4-2] 
[px4-2] ______  __   __    ___ 
[px4-2] | ___ \ \ \ / /   /   |
[px4-2] | |_/ /  \ V /   / /| |
[px4-2] |  __/   /   \  / /_| |
[px4-2] | |     / /^\ \ \___  |
[px4-2] \_|     \/   \/     |_/
[px4-2] 
[px4-2] px4 starting.
[px4-2] 
[px4-2] INFO  [px4] startup script: /bin/sh /home/jaeyoung/vg/vg_ws/install/px4_sim/share/px4_sim/etc/init.d-posix/rcS 0
[px4-2] INFO  [init] found model autostart file as SYS_AUTOSTART=4001
[micro_ros_agent-4] [1673706423.771411] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[micro_ros_agent-4] [1673706423.771543] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[px4-2] INFO  [param] selected parameter default file parameters.bson
[px4-2] INFO  [param] selected parameter backup file parameters_backup.bson
[px4-2]   SYS_AUTOCONFIG: curr: 0 -> new: 1
[px4-2]   SYS_AUTOSTART: curr: 0 -> new: 4001
[px4-2]   CAL_ACC0_ID: curr: 0 -> new: 1310988
[px4-2]   CAL_GYRO0_ID: curr: 0 -> new: 1310988
[px4-2]   CAL_ACC1_ID: curr: 0 -> new: 1310996
[ruby $(which gz) sim-1] I cannot find any available 'gz' command:
[ruby $(which gz) sim-1] 	* Did you install any Gazebo library?
[ruby $(which gz) sim-1] 	* Did you set the GZ_CONFIG_PATH environment variable?
[ruby $(which gz) sim-1] 	    E.g.: export GZ_CONFIG_PATH=$HOME/local/share/gz
[px4-2]   CAL_GYRO1_ID: curr: 0 -> new: 1310996
[ERROR] [ruby $(which gz) sim-1]: process has died [pid 83778, exit code 255, cmd 'ruby $(which gz) sim  -r -v 4 /home/jaeyoung/vg/vg_ws/install/px4_sim/share/px4_sim/worlds/empty_px4_world.sdf --force-version 7'].
[px4-2]   CAL_ACC2_ID: curr: 0 -> new: 1311004
[px4-2]   CAL_GYRO2_ID: curr: 0 -> new: 1311004
[px4-2]   CAL_MAG0_ID: curr: 0 -> new: 197388
[px4-2]   CAL_MAG0_PRIO: curr: -1 -> new: 50
[px4-2]   CAL_MAG1_ID: curr: 0 -> new: 197644
[px4-2]   CAL_MAG1_PRIO: curr: -1 -> new: 50
[px4-2] * SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
[px4-2] * SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
[px4-2] INFO  [dataman] data manager file './dataman' size is 7866640 bytes
[px4-2] INFO  [init] gazebo already running world: 
[px4-2] WARN  [init] PX4_GZ_MODEL_NAME or PX4_GZ_MODEL not set using PX4_SIM_MODEL.
[px4-2] INFO  [gz_bridge] world: -i, model name: x500, simulation model: x500
[px4-2] ERROR [gz_bridge] Service call timed out
[px4-2] ERROR [gz_bridge] Task start failed (-1)
[px4-2] ERROR [init] gz_bridge failed to start
[px4-2] ERROR [px4] Startup script returned with return value: 256
[ERROR] [px4-2]: process has died [pid 83780, exit code 255, cmd 'px4 /tmp/tmp1cmsk87v/ROMFS/px4fmu_common -s /home/jaeyoung/vg/vg_ws/install/px4_sim/share/px4_sim/etc/init.d-posix/rcS -i id0 -d'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[ERROR] [micro_ros_agent-4]: process has died [pid 83786, exit code -2, cmd '/home/jaeyoung/vg/vg_ws/install/micro_ros_agent/lib/micro_ros_agent/micro_ros_agent udp4 -p 8888 --ros-args'].
```

codebot, 2023-02-20:
This should be resolved now; feel free to re-open if a clean "everything" (fresh clone, fresh install of dependencies, etc.) doesn't fix it. Thank you.

closed 2023-02-20 by codebot