osrf/vehicle_gateway #17: PX4 plane doesn't move

Jaeyoung-Lim, 2023-01-14:
**Problem Description**
I have tried out the plane model in the vehicle gateway, but the vehicle is not moving even if the actuators are actuated

[image]

Jaeyoung-Lim, 2023-01-16:
Fixed wing working as a parallel effort in https://github.com/PX4/PX4-Autopilot/pull/20939

codebot, 2023-02-20:
The idea for this repo is a gateway (protocol + API translation) between higher-level software and lower-level autopilots. The goal is to allow the same high-level software to control multiple types of vehicles without code changes. I suppose the simulation stack could also be used for autopilot development, but our current initial goal is towards improving the development experience and portability of high-level software.

We have recently made many changes to the installation instructions now that some required changes have merged in Gazebo and PX4. Following the instructions in the README should now allow a PX4 SITL quadcopter demo to run. Please feel free to re-open if there are issues with that. Cheers

closed 2023-02-20 by codebot