osrf/vehicle_gateway #51: colcon problem

Chenchunxuan, 2023-02-25:
Updated 0 paths from the index
patching file src/main/target/SITL/target.h
Hunk #1 FAILED at 143.
1 out of 1 hunk FAILED -- saving rejects to file src/main/target/SITL/target.h.rej
gmake[2]: *** [CMakeFiles/betaflight-firmware.dir/build.make:119: betaflight_sim/src/betaflight-firmware-stamp/betaflight-firmware-patch] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/betaflight-firmware.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2

when I run "colcon build --event-handlers console_direct+" ,I meet this problem , I think I am  not alone ,please help.

Chenchunxuan, 2023-02-25:
only betaflight_sim package can't get success.

ahcorde, 2023-02-27:
Can you share the full trace ? I don´t have enough information to help you.

codebot, 2023-03-03:
Thank you for this report. It should be fixed now that #31  is merged. Feel free to re-open if the issue remains, after a clean rebuild and installing the dependencies via `rosdep` in the installation instructions.

closed 2023-03-03 by codebot