osrf/vehicle_gateway #98: Cannot create file .../microdds_client/dds_topics.yaml

audrow, 2023-05-04:
I was getting this error locally before seeing it the nightly: 
https://pipelines.actions.githubusercontent.com/serviceHosts/660d4560-3641-47a4-9618-bec6301420b8/_apis/pipelines/1/runs/813/signedlogcontent/2?urlExpires=2023-05-04T19%3A43%3A38.2952873Z&urlSigningMethod=HMACV1&urlSignature=9znXE7sIjIjOAPjZLXyQKM%2B5ANeokkOcLVWEUcBOMLg%3D

It seems that the `microdds_client` folder no longer exists in PX4. I wonder where it went... (will be looking into it).

audrow, 2023-05-04:
It'd be really nice to pin the version of PX4 that we use.

audrow, 2023-05-04:
I think this is the offending commit: https://github.com/PX4/PX4-Autopilot/commit/9e5420bbbd0c3fe2001ae4a1ebe6c6c3abec830c

codebot, 2023-05-04:
Oh! Thank you for chasing this down.

codebot, 2023-05-04:
Even if we pin the PX4 version, I think the submodule versions will be tracking forward, so it actually might make it less stable since they may be moving in lockstep sometimes (like this example, potentially).

PR #99 (2023-05-04): Adjust to rename
Fixes #98
closed 2023-05-04 by audrow