osrf/vehicle_gateway #127: colcon build fail

brewfalconenterprises, 2023-09-17:
Using the provided build instructions, received the following output at the `colcon build` step.
```
Summary: 17 packages finished [2min 11s]
  1 package failed: betaflight_sim
  4 packages aborted: px4_msgs px4_sim ros_gz_image zenohc
  13 packages had stderr output: betaflight_gazebo betaflight_sim micro_ros_agent micro_ros_msgs px4_msgs px4_sim ros_gz_bridge ros_gz_interfaces ros_gz_sim rpyutils vehicle_gateway_betaflight vehicle_gateway_python_helpers zenohc
  8 packages not processed
```

Tuxliri, 2023-11-29:
I also have the same issue, albeit with a different failing package:

```
Summary: 18 packages finished [15min 53s]
  1 package failed: vehicle_gateway_multi
  4 packages aborted: betaflight_sim px4_sim ros_gz_image vehicle_gateway_px4
  10 packages had stderr output: betaflight_gazebo betaflight_sim micro_ros_agent px4_sim ros_gz_sim vehicle_gateway_betaflight vehicle_gateway_multi vehicle_gateway_px4 vehicle_gateway_python_helpers zenohc
  6 packages not processed
```
Did you manage to find a fix?

brewfalconenterprises, 2023-11-30:
I built the failing packages one at a time and got most successfully built. There were still a few that failed but were unnecessary for my work.Sent from my Verizon, Samsung Galaxy smartphone

Tuxliri, 2023-11-30:
Yes I found a bug with the version of `px4_msgs` that was being used, see #129.

ahcorde, 2023-12-04:
Feel free to reopen the issue if the problem persists

closed 2023-12-04 by ahcorde

khunglong207, 2024-02-25:
@ahcorde I also have the same issue, albeit with a different failing package:
CMake Error at CMakeLists.txt:25 (find_package):
  By not providing "Findgz-transport12.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "gz-transport12", but CMake did not find one.

  Could not find a package configuration file provided by "gz-transport12"
  with any of the following names:

    gz-transport12Config.cmake
    gz-transport12-config.cmake

  Add the installation prefix of "gz-transport12" to CMAKE_PREFIX_PATH or set
  "gz-transport12_DIR" to a directory containing one of the above files.  If
  "gz-transport12" provides a separate development package or SDK, be sure it
  has been installed.
Failed   <<< ros_gz_sim [2.36s, exited with code 1]
-- Installing: /home/theshy/vg/install/ros_gz_interfaces/local/lib/python3.10/dist-packages/ros_gz_interfaces/ros_gz_interfaces_s__rosidl_typesupport_fastrtps_c.cpython-310-x86_64-linux-gnu.so
Summary: 4 packages finished [7.70s]
  1 package failed: ros_gz_sim
  7 packages aborted: betaflight_configurator betaflight_controller micro_ros_msgs px4_msgs qgroundcontrol ros_gz_interfaces vehicle_gateway_python_helpers
  1 package had stderr output: ros_gz_sim
  17 packages not processed
  
Can u help pls